CrashOverride85/zc95 #1: Big Thanks & Audio

seuchato, 2022-01-24:
Hi
1st: Big fat thanks for developing this! Excellent work, indeed!
I ordered the stuff and expect it to arrive at my home soon. I'd be very interested in adding an audio input. Is there something else than the extra daughter board to be considered when assembling the pcb?
Do you have an eta for the audio daaghter board?
I'd like to suggest to add the audio jack seperate from the serial jack. I think it would be nice to have both. 
Greez
seuchato

CrashOverride85, 2022-01-24:
Hi, I'm not sure about the audio input yet, it's on my list, but if I'm honest I'd say not soon, it's more an idea at this point.

I agree it'd be nice to have both audio and serial, but adding an audio jack would mean new versions of the front panel and main board, in which case it'd probably make more sense to just integrate the audio input circuitry straight onto the main board (there's plenty of space), and forget about a daughterboard.

The reason for adding the possibility of a daughterboard in the future was so audio input could to be added to an existing box, rather than having to build another.

bkifft, 2022-04-22:
Does it actually need analog audio in? Shouldn't the pi's  be beefy enough, to parse mp3 straight to output? Skipping the DAC and ADC.

If someone can describe me the input output behaviour I can try to do a proof of concept.

CrashOverride85, 2022-04-22:
Need? not really really, it's very much on my nice-to-have list :)
It's using Pico's not Pi's, which I don't think are really powerful enough to decode MP3 in any practical way - although a quick google suggests it is possible if that's all it's doing and with very low bitrate MP3s.
Getting MP3's into it would also be awkward... in theory there's a micro SD slot on the back of the display module, but I can't remember if it's connected, and even if it is, having to open the box to change the MP3's wouldn't be great.

I'll be honest, an internal digital source isn't something I'd considered, though. I've only ever used the mic audio input on a mk312bt, and that was just to trigger off, ummm, noises, made by the person attached to it, rather than dedicated estim audio files.

>If someone can describe me the input output behaviour I can try to do a proof of concept.
Heh, yes, I think figuring out that behaviour is likely going to be the hardest part, and a large part of the reason I've not attempted this yet.

CrashOverride85, 2022-08-27:
Audio input board and first pattern using it now added (#13). The current pattern is only really suitable for microphone input, but I hope to add a couple more over the next few months. 
Its using the existing hardware for the time being, so no extra sockets, etc. on the front panel.

closed 2022-08-27 by CrashOverride85