CrashOverride85/zc95 #2: Deviation kicad-gerber

electric1711, 2022-02-04:
Hi

Thanks for the project.
I'm in the process of building one.
I noticed that the Gerber flies for the frontpanel are available in version 0.2 the bom list and kikad files are in version 0.1.
Here is the question, are still all unmarked resistors 10k except those marked with 100R.
Is U2 still a PCF8574 and what is the replacement for U1.
What are the holes above the POTs for?

CrashOverride85, 2022-02-04:
Hi,

I am so sorry... I seem to have uploaded completely the wrong gerber for that board :(
It was for an alternate version of the front panel that used 4 rotary encoders for the channel power adjustments instead of 4 POTs. 
I have corrected this, and uploaded the correct gerber. 
Well spotted, and thanks for letting me know - I can only apologise if you've already ordered the board :/

BTW, which holes did you mean?

Crash

electric1711, 2022-02-04:
Hi

Thanks for the quick reply
Fortunately, I noticed it before ordering.
The additionally holes would probably have been for other pots (v0.2) .

commit b2d21ae2 (2022-02-04): fixes #2 - wrong gerber file for front panel (back) PCB
commit fc7bbaea (2022-02-04): fixes #2 - wrong gerber file for front panel (back) PCB
closed 2022-02-04 by CrashOverride85