CrashOverride85/zc95 #3: Polyfuse is undersized

CrashOverride85, 2022-04-23:
The current Polyfuse is a BK60-050 with a trip current somewhere between 0.5A and 1.0A, which is a bit too low. In TENs mode, when set to a large pulse width and high frequency with all the channels set to max, the box can pull ~900mA, which is enough to trip the fuse. 
The schematic calls for a 1A fuse, so a BK60-090 (0.9A - 1.8A) or BK60-110 (1.1A - 2.2A) is probably a better option.

heavyrubberslave, 2024-09-01:
Would this be the recommended replacement: https://www.lcsc.com/product-detail/Resettable-Fuses_Brightking-BK60-110-DI-E0-6_C523033.html

CrashOverride85, 2024-09-01:
Yes, that should be fine. 
TBH I'd probably go for the lower rated [BK60-090](https://www.lcsc.com/product-detail/Resettable-Fuses_Brightking-BK60-090-DI-E0-5_C523029.html) instead, as in my experience so far, it's only possible to trip the current fuse if you're deliberately trying to, so I think the BK60-110 is likely higher that it needs to be.