CrashOverride85/zc95 #11: Possible DAC alternative?

shiny-mira, 2022-07-21:
Hi,

first - let me say thank you - this is a really great project and I am looking forward to try to build this.

I noticed that the DAC that is used in this (the MCP4728) is currently completely unavailable; Mouser, e.g., gives an estimated in-stock date of December 2023. I was just wondering if you randomly know of another DAC that one might be able to use instead - or if I just have to wait it out.

Thank you again :)

CrashOverride85, 2022-07-22:
Ahhh, that sucks, I hadn't realised. No - I'm not aware of anything else that could be used as a drop in replacement that's available :(
If I can find another suitable DAC that's still got plenty of stock in most places, I might have a go a tweaking the design to use that - having footprints for both may be an option

It does look like the MCP4728 is still in stock at conrad (I've never bought anything from them before though).

CrashOverride85, 2022-10-13:
It looks like the MCP4728 is currently in stock at LCSC, although at a somewhat inflated price. I guess an alternative is to buy a board with it on and remove it - e.g. the Adafruit ADA4470, which is about the same as the part costs at LCSC anyway.
Hopefully the parts situation will improve early next year...

closed 2022-10-15 by CrashOverride85