CrashOverride85/zc95 #27: Problems with the adjustment knob

piepa12, 2023-05-21:
I also built myself a box and I think the concept and the idea are great!

I have a problem with the adjustment knob. I have to turn it very slowly so that the values change. Is the problem known or did I install the wrong component? 

Greetings from Germany.

CrashOverride85, 2023-05-22:
hmmm, how slowly? You can't turn it that fast, but it shouldn't be much worse than, for example, the adjust dial on a 2B (if you have or have used one)

piepa12, 2023-05-22:
I made a short video of the mistake.

[Video](https://youtu.be/Rgsg2NaYpd8)

CrashOverride85, 2023-05-22:
Thanks, yeah, something is definitely wrong there, it should be possible to adjust it a lot faster than that.
Is it the same rotary encoder from the BoM, or if not, could you let me know the part number? 

But I think it's most likely being caused by the interrupt pin of U2 (pin 13) on the front panel not being connected to the Pico (pin 9) on the main board, could you try measuring for continuity between the two?

If there isn't continuity, I'd suggest checking the cable between the main board on the front panel - the interrupt is on pin 2.

closed 2023-07-31 by CrashOverride85