CrashOverride85/zc95 #37: lua upload issue

RupertB-ES, 2023-08-02:
Error uploading lua script - syntax appears to be clean according to https://www.jdoodle.com/execute-lua-online/
ps Github doesn't support attaching lua files either !

C:\zc95>python lua_upload.py --script ./lua/ramp_exp_phases_rb.lua --ip 192.168.1.157 --index 3
Connecting
Connection opened
Uploading...
Didn't get any message, expected Ack (msgId = 261)
[ramp_exp_phases_rb.txt](https://github.com/CrashOverride85/zc95/files/12237843/ramp_exp_phases_rb.txt)

Websocket connection closed
Failed

C:\zc95>python lua_upload.py --script ./lua/ramp_exp_rb.lua --ip 192.168.1.157 --index 3
Connecting
Connection opened
Uploading...
Done!
Websocket connection closed

CrashOverride85, 2023-08-02:
Hmm, looks like it's hitting a line length limit. At the moment, each line is limited to ~155 characters. If you break up the 3 long lines with line breaks, it should upload ok.

I'll look at increasing this limit, and also include an error when whatever the limit becomes is reached.

RupertB-ES, 2023-08-02:
Thanks 
I simplified the logic and it fixed the problem pro tem
Also the debug print statement seems to have a real hit on performance

CrashOverride85, 2023-08-03:
Yeah, I don't think there's much I can realistically do about that one :(
I'd suggest using print() really is only good for debugging - Loop() is called often enough that having it call print will badly affect it, to the point it may start dropping debug messages.

CrashOverride85, 2023-08-12:
Should now be resolved in v1.6 by increasing max line length limit, and showing an error then the new limit is hit

closed 2023-08-12 by CrashOverride85