CrashOverride85/zc95 #41: Issue on script startup

RupertB-ES, 2023-08-06:
Got this trace on starting a pattern - its worked ok up to now and I hadn't changed it at all

C:\zc95>python pattern_gui.py --ip 192.168.1.154 --index 3
Connecting
Connection opened
Didn't get any message, expected PatternDetail (msgId = 2)
Traceback (most recent call last):
  File "C:\zc95\pattern_gui.py", line 220, in <module>
    gui = ZcPatternGui(root, pattern, zc_messages, rcv_queue, args.debug)
  File "C:\zc95\pattern_gui.py", line 24, in __init__
    self.InitDisplay(self.root)
  File "C:\zc95\pattern_gui.py", line 76, in InitDisplay
    self.DrawPatternFrame(root, row=0, col=0)
  File "C:\zc95\pattern_gui.py", line 92, in DrawPatternFrame
    Label(pattern_frame, text=pattern["Name"], font='Helvetica 18 bold').grid(row=0, column=0, padx=5, pady=5)
TypeError: 'NoneType' object is not subscriptable

RupertB-ES, 2023-08-06:
PS Worked ok the next time I ran the same command

RupertB-ES, 2023-08-08:
Its only done it the once - I notice that the speed of running a remote pattern via python seems a bit variable especially initially  launching the gui

CrashOverride85, 2023-08-09:
hmmm, interesting, thanks. How fast the pattern actually runs _shouldn't_ be affected by running remotely, even if the GUI runs a bit slow

RupertB-ES, 2023-08-09:
I think its just the speed of the python vm launch/run but its surprising its so variable as its on an i7 box with nothing else much going on

CrashOverride85, 2023-08-10:
I wonder if it's a wifi thing / not a great connection for some reason. It shouldn't me taking long for the gui to start up

RupertB-ES, 2023-08-10:
I am seeing some WiFi strangeness but haven't figured out a pattern - it seems to be something DHCP Lease related - other wireless devices I have seem to be ok - phones/Echo Dots.
The connection drops after several hours sometimes a couple of days and doesn't reestablish unless I power cycle the box  - just selecting reconnect gives me a message Connected - No IP most times

RupertB-ES, 2023-08-11:
Similar error today in pattern list after rebooting box
5>python pattern_list.py --ip 192.168.1.146
Connecting
Connection opened
Didn't get any message, expected PatternList (msgId = 2)
Traceback (most recent call last):
  File "C:\zc95\pattern_list.py", line 22, in <module>
    patterns = zc_messages.GetPatterns()
  File "C:\zc95\lib\ZcMessages.py", line 52, in GetPatterns
    return response["Patterns"]
TypeError: 'NoneType' object is not subscriptable

CrashOverride85, 2023-08-11:
Thanks - I may have seen the DHCP thing when fairly rapidly rebooting the box, and figured it was probably just the router. I'll take a closer look at that

CrashOverride85, 2023-08-13:
I'm starting to wonder if WiFi flakiness is related to #46  and #25

RupertB-ES, 2023-08-13:
Just reflashed the box this morning - it feels snappier but have not tried anything in anger on it