CrashOverride85/zc95 #46: Wifi Config AP mode fails

jlatex, 2023-08-13:
With 1.6 release and with current git head, after selecting 'config wifi/ap mode' on the UI the UI freezes and the box is unresponsive.  Serial log:

```
CMenuApMode()
SetupWebInterface()
Loop time: 514, batt: 100
CAnalogueCapture::stop()
CAnalogueCapture::stop() done
CMenuApMode::wifi_scan(): Start scanning for wifi networks
CWifi::init(): wifi/cyw43 init success
dhcp_release_and_stop()
```

Since this is a new box I've not had this working yet

jlatex, 2023-08-13:
Oh, this might be https://github.com/CrashOverride85/zc95/discussions/25 I'll try that!

jlatex, 2023-08-13:
Okay, this is #25 and our friends box has the same problem, so suggest adding a note to the discussion from the docs for now until there's a solution.  Updated title of this issue

CrashOverride85, 2023-08-13:
Thanks for reporting this - I'm going to treat this more as a bug in the design (poor choice of MOSFET as you've identified), than just a documentation issue. Although short term, I agree that's a good idea.

CrashOverride85, 2023-08-13:
Added to known issues, will look for a more suitable mosfet

jlatex, 2023-08-15:
[image]

CrashOverride85, 2023-08-15:
Ahh cool, good to know it works, thanks for letting me know. I'm going to try a smaller diode when it arrives, and update BoM & build notes to match

closed 2023-08-26 by CrashOverride85