CrashOverride85/zc95 #47: Add quick 'get starting building the source code' notes

jlatex, 2023-08-13:
I'd not built anything for rp2040 before, so would save a little time if there was a note somewhere of the quick steps to get up and running 

i.e.

- checkout repo, ensure recursive (if you already have it checkout out then make sure to update the modules example 'git submodule update --recursive')
- get the pico-sdk (example 'git clone https://github.com/raspberrypi/pico-sdk.git; cd pico-sdk/; git -C . submodule update --init; export PICO_SDK_PATH=`pwd`' )
- get the arm gnu toolchain such as using https://lindevs.com/install-arm-gnu-toolchain-on-ubuntu and ensure it's on your path (example 'export PATH=$PATH:`pwd`/gcc-arm-none-eabi/bin/')
- in the zc95/source/zc95 directory 'cmake -Wno-dev .' then 'make' as usual

closed 2023-08-13 by CrashOverride85

jlatex, 2023-08-14:
Thanks btw, got it working with jlink/vscode too which is great for debugging (although as far as I can tell you have to have the serial port connected as well to see the debug output)

CrashOverride85, 2023-08-14:
Ahhh cool :+1: , it makes life so much easier.
And yeah, still need to use a serial cable to get the debug output