CrashOverride85/zc95 #48: Consider adding a serial baud rate menu config option

jlatex, 2023-08-13:
By default the serial port runs at PICO_DEFAULT_UART_BAUD_RATE (115200).  My longish serial cable and usb serial adapter on aux couldn't keep up with that and so I edited CDebugOutput.cpp to drop the baud rate to 38400.  Ideally this would be a menu config option.

CrashOverride85, 2023-08-13:
Depending on what you're using serial for, this might be slightly harder than it seems.
For the new remote control & Lua upload over serial stuff, having it changeable via the menus should be ok.

But for debug/diagnostic output, the main issue I can see is that on startup there's a few things it does before reading the EEPROM, so it's going to be outputting at the default baud rate until that happens. Switching baud rates during startup doesn't seem ideal really.

jlatex, 2023-08-13:
That's fair; I don't mind keeping mine at 57600 or something by just compiling it myself.  And if anyone else has issues with the speed perhaps consider using that as the default.

closed 2023-08-13 by jlatex