CrashOverride85/zc95 #52: Wishlist item, display power levels

jlatex, 2023-08-14:
While using a zc95 we miss being able to see the power level as a percentage. This is handy when setting a level similar to one used before (knowing that say a 57 on channel A is a 'sweet spot'), but also when adjusting the control (wanting to turn it up by 1%). But the display doesn't have a lot of space for it with the current UI.

This issue is a wishlist for the future to find a nice way to display power levels.  

I did a quick hack that overwrites the screen with the levels of the four channels whenever you change one of the levels (vanishing again after half a second ish) and that works okay, but it looks a bit messy (and hagl doesn't have a nice scaled large font so it just scales the bitmaps).  It's here, but not as a PR as it's probably not for everyone.

https://github.com/jlatex/zc95/commit/838931ccb7025896764dbabc9e9813d6751eede1

CrashOverride85, 2023-08-14:
Interesting idea, probably wouldn't hurt to have as an optional thing that can be turned on/off as wanted

CrashOverride85, 2023-08-16:
Hi, yeah, I'm also crashoverride85 on discord. 
Speaking of discord, I'd recommend joining [Joanne's E-Stim Community](https://discord.gg/rY8C27S) discord server, I kinda assumed that's how you found the project. There's a few people in there particularity knowledgeable about estim box design - the #boxes-pulse-based-diy channel is especially relevant to this kind of project

jlatex, 2023-08-22:
Tested, looks great to me!  Thanks!

closed 2023-08-22 by jlatex