CrashOverride85/zc95 #53: Problem with lua_upload

RupertB-ES, 2023-08-17:
Downloaded the latest  lua_upload.py for  rel 1.6 and am getting

python lua_upload.py --script ./lua/ramp_exp_phases_rb.lua --ip 192.168.1.146 --index 1
Traceback (most recent call last):
  File "C:\zc95\lua_upload.py", line 9, in <module>
    from lib.ZcSerial import ZcSerial
  File "C:\zc95\lib\ZcSerial.py", line 2, in <module>
    from serial import Serial
ImportError: cannot import name 'Serial' from 'serial' (C:\Python310\lib\site-packages\serial\__init__.py)

CrashOverride85, 2023-08-17:
I've not seen that before, but it looks like an issue with the pyserial package. I've not tried it, but but [this](https://stackoverflow.com/questions/67901323/dont-understand-reason-for-import-error-cannot-import-name-serial-from-seri#comment120017764_67901401) seems to suggest trying:
`pip install --force-reinstall pyserial`

RupertB-ES, 2023-08-17:
Cracked it - thanks - ps menu grouping working nicely :-)

CrashOverride85, 2023-08-17:
Excellent, glad it's working 🙂

closed 2023-08-17 by CrashOverride85