CrashOverride85/zc95 #56: MQTT/XToys integration

RupertB-ES, 2023-08-17:
I believe there is a LUA implementation of MQTT -any plans on this or other routes to integrate with XToys?

RupertB-ES, 2023-08-18:
This is what caught my eye https://github.com/xHasKx/luamqtt

CrashOverride85, 2023-08-18:
I think the problem's going to be that it depends on luasocket, which depends on a BSD-style sockets API, which LwIP used by the Pico SDK doesn't really provide. Plus I think trying to run Lua scripts of that size/complexity just isn't going to work well (if at all).
I think the way to get mqtt support in Lua would be to add a wrapper around the LwIP MQTT client to expose it in the Lua environment. 

Also I need to have a think about how it would work when running Lua script locally, as at present, the box will only connect to WiFi when in remote access mode.

RupertB-ES, 2023-08-19:
I guess there are 2 routes - initially could the LUA client just handshake with a python app and let python do the work but in the longer term having the LUA client "phone home" and start the wifi in code would be the better way performance headroom permitting.

CrashOverride85, 2023-08-20:
Yeah, doing _something_ via the python app might be an option too