CrashOverride85/zc95 #77: PIO program question.

martinmikula, 2023-10-25:
Hi,

at first thanks for great project... 
I'm not very good in pio programming in RP2040, but i found one thing which doesn't make sense for me:

you pulling block - 32bits into OSR register
`pull block`
then shifting 2x8bit into x and y regs
`out x, 8
out y, 8`
and then shifting 32bits into null
`out null, 32`

How the PIO works if there is not enough data to shift? I found that it can be configured to auto pull, but i think this is not the case, otherwise the x and y will be messed up... Probably you will also find if there is a problem, I just not understand so writing for clarification. :)

BR
Martin

CrashOverride85, 2023-10-26:
I've only written two PIO programs, and that was a couple of years ago now... so I can't quite remember unfortunately 😥️
I _think_ the reasoning behind the shift was just to make sure any remaining data got discarded, as only the first two bytes have any meaning.
When the `pull block` runs, it grabs the next 32bit entry on the FIFO buffer.
The code seems to do what it needs to, which is why I've not looked at it for so long

martinmikula, 2023-11-02:
Ok, than it's fine... :) I'm also programmer and hw designer, so if you have any new ideas or just need some help with some new features or problems please let me know... :) I don't have the hw now, but i'm working on it... :)

CrashOverride85, 2023-11-04:
heh thanks :)
No major changes planned for the hardware in the short term, but always trying to add bits to the firmware :slightly_smiling_face: 
If you get one built, I'd deffo be interested in hearing how you get on with it, especially if it "just works", I only tend to hear about the ones that don't lol

closed 2023-12-13 by CrashOverride85