CrashOverride85/zc95 #111: 433 mhz collar beeping only every other time unless quickly repeated

electro991, 2024-07-30:
It seems that the 433mhz collar i have requires the zc95 to send the beep signal twice every time it is required

if only sent once, for some reason the collar will skip every other beep request from the zc95

this behaviour is not the case for shock or vibrate mode

it is especially problematic if using shock choice as a beep is only heard every other time a choice needs to be made

electro991, 2024-07-30:
Correction; it seems to be inconsistent for vibrate as well. I believe the original remote sends its signal 2x as the vibration is longer on original (2x the length)

i believe it may just be that shock choice is just too brief for the collar to work every time. In other modes for example Round Robin, it never skips a beat at 1000ms

CrashOverride85, 2024-08-01:
I've tweaked the times so it should now be sending sending 2 sets of messages to a collar instead of 1 set as before.
This isn't a problem I've experienced, so I can't really test it beyond that it still works for me,
I've attached an updated F/W for the zc95 that has this fix, and the one for needing a reboot after changing settings, could you let me know if it works any better?
[zc95.uf2.zip](https://github.com/user-attachments/files/16460957/zc95.uf2.zip)

electro991, 2024-08-02:
Awesome. Thank you. I will test after the weekend and get back to you.

electro991, 2024-08-07:
Good evening,
I have tested this change and it is better but still missing some warnings on chan4 at times. I have loaded the increase_shockchoice_pulse_length branch and increased simple_channel_pulse(3, 600) to simple_channel_pulse(3, 1200) and recompiled. With that, it appears to be consistently working for me. I came up with 1200 by testing at what point it became consistent using the roundrobin pattern and increasing the delay until it worked every time.

CrashOverride85, 2024-08-08:
I'm assuming it's down to slightly different versions of the collars. I'll update it to 1200 shorty, thanks for experimenting that letting me know what works

electro991, 2024-08-08:
Thank you. Much appreciated. I have tested with 3 different of that model too. Pretty confident about the value

CrashOverride85, 2024-08-17:
Updated to 1200. With the collar/transmitter I've got, that consistently results in 2x pairs of beeps rather than one as before, but that's probably better than being intermittent with some combinations

closed 2024-08-17 by CrashOverride85