CrashOverride85/zc95 #117: LCD startup

electro991, 2024-08-17:
Nothing serious but i have the following odd thing happening with the LCD

on assembled unit 1, screen on power up goes all white for a seconds, then black, then 2-3 seconds later, the populated main screen appears. This is rather clean except the white for one second

on assembled unit 2, screen on power goes white for a second, then mainly white with somewhat garbage pixels of purpleish colors for about another second, then main menu unpopulated appears, then about 2 seconds later main menu populates. This behaviour I think is not so clean

After troubleshooting, the problem follows the screen (keeping everything else consistent, including cable)

i tried a 3rd screen i had lying around and it behaves like assembled unit 2

i am not sure if within code the init of the screen can be optimized in any way. I have seen some discuss this behaviour on doing a google search. Turning off the backlight thru code during init seems to do it for them. I have not inspected the code for this project yet to determine if this has been tried.

CrashOverride85, 2024-08-17:
Mine behaves as per your unit 1, and that white flash is essentially design flaw in the zc95 main board unfortunately. The backlight pin on the display is connected to a PCF8574 I/O expander, which on power up defaults all the I/O pins to high. The brief flash is the time it takes to do some initial init then set that output pin to low. It then switches it back on when init is complete.

There's probably some scope to init the I/O expander and switch off the backlight slightly sooner, but I think there'd still be a white flash, and it wouldn't do anything for the other displays.

It sounds like the backlight pin isn't working on the other two displays, although it also seems quite unlikely that two displays would have a fault. I'm struggling to spot what could be causing it though.

Having tested mine, it behaves as per your unit 2 if I comment out line 192 in zc95.cpp

electro991, 2024-08-17:
Thank you for the insights. I will check that pin’s solder and continuity to the pcf and get back in a few days.

electro991, 2024-09-04:
Fixed! thank you.  It was a loose contact on the cable. It was on and off. For some reason it worked on one screen lol. Just had the right angle for the contact to work.

closed 2024-09-04 by electro991