CrashOverride85/zc95 #119: HELP: JLCPCB issues ⁉️

moarkaboom, 2024-09-01:
After placing my order with JLCPCB they sent me the following response.
Even though I am not entirely sure what option A is, it seems the preferable one. I am unsure if they will recommend suitable parts or if I have to redefine them myself. 
It's the first time for me ordering a populated PCB and my electronics skills are *very* basic. It would be a tremendous help, if somebody could support me in this process.

> 
Hi there,
So sorry to bother you
Sorry to bother you, but there is one thing that we want to confirm with you about your PCBA order  SMT02408311921237_Y17 8313320A before proceeding.
As shown below, seems the packages of the parts (C149504、R59、C8、C49678、C28323、C4310、C4310、C17414 ) cannot match the corresponding pads on the PCB board, so we are unable to assemble them.
For this case there are 3 options for your choose as follows:
A- we enable the replace part button for you that you can re-select the corrected part
B- we leave this parts unpopulated and refund cost of the mismatch parts back to you to proceed and can’t ship the unsoldered parts with PCB due to the customs clearance issue,
C-cancel this PCB order and SMT order to place new order (there is no replace gerber file button for SMT order)
Pls consider and advise which case suits you, A or B or C ??           
Your early reply will be highly appreciated, thank you so much!
 
[image]

 
[image]

CrashOverride85, 2024-09-01:
arggg thanks for reporting this...
It looks like I've got the wrong part (0805 vs 0603) in the BoM/CPL files for those parts. I've been back and checked, and it was the same wrong parts when I ordered it from JLCPCB, I have no idea how they managed to assemble mine!

I've about to push a fix to the BoM & CPL files, so you can either do option C and use that, or option A and make these substitutions:
C4310   => C22843 (1.5k)
C17798  => C23228 (680)
C49678  => C14663 (100nF)
C149504 => C25803 (100k)
C17414  => C25804 (10k)
C17470  => C22807 (150k)
C28323  => C15849 (1uF)

Sorry about that!

electro991, 2024-09-01:
I had the same when i ordered but they advised they could still solder so that’s the option i took. Thanks for the fix

moarkaboom, 2024-09-02:
Thanks for taking care of this so quickly!
I noticed you didnt replace C15850, does that mean this part remains?

CrashOverride85, 2024-09-02:
Ahh yeah good spot, I missed that one
Now fixed
C15850 => C96446

closed 2024-09-10 by moarkaboom