CrashOverride85/zc95 #121: Sourcing help: 0.5 Ohm resistor

moarkaboom, 2024-09-10:
I am having real trouble finding the correct part for C479296 which is a 0.5 Ohm Resistor used on the output board (R1, R26).
Can somebody either point so a supplier (preferably EU) or to a better available part?

CrashOverride85, 2024-09-10:
Hadn't noticed that was out of stock, I'll update it.

From LCSC, [C479327](https://www.lcsc.com/product-detail/Through-Hole-Resistors_VO-MF1W-0-5-2-6T63_C479327.html) would be a suitable replacement, or [MCKNP01SJ050KA10](https://uk.farnell.com/multicomp/mcknp01sj050ka10/res-0r5-5-1w-axial-wirewound/dp/1903704) from farnell.

The resistor is listed in the BoM file, and it seems C479296 is still in stock with jlcpcb, so if it wasn't un-ticked when ordering, jlcpcb would have assembled it. The main reason some parts are marked as hand assembly is because it is (or was, not checked recently) quite a bit cheaper.

moarkaboom, 2024-09-10:
Thanks  for the response. Bur I don't need to source it now, right?

CrashOverride85, 2024-09-10:
Nope :)

closed 2024-09-10 by moarkaboom