CrashOverride85/zc95 #123: Suggestions for improvements/changes

heavyrubberslave, 2024-10-18:
Hello,

I successfully built the ZC95. Thank you so much for all the hard work you put into this and for sharing it with the public.

I have a list of things I see some improvements could be made. You may or may not had reasons for solving it the way you did. Let me post the list here and I can do separate issues in case you think it's worth looking into some of those:

### FrontPanel Controls
* J5 comes with just a pin header instead of a socket if assembly by JLCPCB is chosen. (Could be removed from BOM.)
* Make it possible to use a 2x2pin (Dupont or alike) plugs for the 4 LED buttons so the button's wires don't need to be soldered onto the board. 
* Replace the Adafruit display with a cheaper alternative (with same driver).
* Use a rotary encoder which has the same/similar length as the potentiometers.

### Mainboard
* Use 3.5mm audio jacks for channels as well (for compatibility with MK312/E-Stim System's 2B)
* Use a 10pin JST XH pin header/socker for display (same on the display end), that way one could use standard 10pin JST XH cables
* Use a (15V PB) USB-C socket for charging

### Audio board
* Add option to use the assembly service for U1 (and potentially other components)

CrashOverride85, 2024-10-18:
Hi, glad to hear you've built it, hope you have fun with it :)

Thanks for your feedback, these are my comments on the issues:

### FrontPanel Controls
> J5 comes with just a pin header instead of a socket if assembly by JLCPCB is chosen. (Could be removed from BOM.)

Ahh yeah, that was unintentional - I think I must have unticked that part when ordering, and forget to take it out of the BoM file. Probably also means the LCSC part number is wrong for part in the schematic if it got populated with a pin header not a socket.
I'd always intended for that to be hand soldered.

> Make it possible to use a 2x2pin (Dupont or alike) plugs for the 4 LED buttons so the button's wires don't need to be soldered onto the board.

Didn't think of that, good idea, just changing the pin spacing to a 2x2 2.54mm grid should do it, then it could either be directly soldered or a connector used

> Replace the Adafruit display with a cheaper alternative (with same driver).

I chose that specific display because at the time it seemed to be widely available. A lot of the cheaper displays often either don't have an easy to search for model name, were only available from one retailer, and/or looked like the sort of part that would disappear quickly. I also wanted a complete module I could mount to the front panel (i.e. not just a bare display) for simplicity's sake.
The ADA358 module also had great documentation that actually gives dimensions / positions for mounting holes, it's surprising how few other display modules have this available!

Also whilst the display driver does support a wide variety of displays, it's a compile-time configuration thing, and I really don't want to get into maintaining two separate firmware binaries unless I have to (e.g. if the current display gets discontinued), so I'm not planning to change the display.

> Use a rotary encoder which has the same/similar length as the potentiometers.

Interesting, hadn't spotted there was much of a difference, I'll see what I can find

### Mainboard
> Use 3.5mm audio jacks for channels as well (for compatibility with MK312/E-Stim System's 2B)

This was a very deliberate choice as most of my cables are 2.5mm, and I've got 2.5mm<>3.5mm adaptors for things that aren't. I think @shiny-mira has a fork with a modified version of the main PCB and front panel which does have 3.5mm connectors though

> Use a 10pin JST XH pin header/socker for display (same on the display end), that way one could use standard 10pin JST XH cables

This really comes down to the choice of display :/ JST XH connectors have a 2.5mm pin pitch, but the header on the ADA358 display is 2.54mm, so it wouldn't be possible use a ready made JST cable. I guess cutting the end off one and soldering to the display would be an option, as it only really needs to be removable at one end. That would at least remove the need for a crimp tool to create the cable.

> Use a (15V PB) USB-C socket for charging

I think all USB-C PD IC's are SMT, and I was trying to keep the main board through hole only. I'm not keen on using a separate USB-C PD module as selecting one that's likely to still be easy to find in a year or two is probably tricky, and mounting it could be awkward.
Very long term (> 1 year definitely), I do plan to combine the main / output / audio boards into one SMD board, swap over to a lithium battery, and move to USB-C charging. But I agree a 15v barrel jack is far from ideal.
All I can suggest for the moment is a 15v USB-C to barrel jack cable - that's what I've been using and works well enough.

### Audio board
> Add option to use the assembly service for U1 (and potentially other components)

Redesigning this board to be SMT and using the JLC assembly service is high up my todo list - and definitely for more than just U1. I'll probably leave just the pin sockets and _maybe_ the relay requiring hand soldering.

> I think the pin headers J1 and J2 are not included in the BOM

It's line 9, part C35445 on the audio board tab of the BoM spreadsheet 

> I think the IC socket for U2 (2x7pin) is not included in the BOM

ahhh good spot, looks like the socket for U3 is also missing, but you likely had a spare from the main board

electro991, 2024-10-21:
Regarding the 2.5mm plugs, i used this adapter. I like it as it prevents wear and tear on the sockets on the units itself 

2.5mm Male to 3.5mm Female Aux Stereo Cable Connector Adapter
https://a.aliexpress.com/_ms5xhnO

electro991, 2024-10-21:
I had no issues here with this adapter