CrashOverride85/zc95 #125: Audio board rework

CrashOverride85, 2024-10-22:
Whilst it does mostly work, there are numerous issues with the current audio board:
* Silkscreen for D3 is backwards
* 2n7000 is poor choice of MOSFET (see #114)
* R27 pull down should either be a pull up, or just omitted
* BoM spreadsheet is missing IC sockets for U2 and U3
* Having that one SMD part in there isn't great

Solution: redesign the board to be fully SMD, using the the JLCPCB assembly service. Swap to AO3400 MOSFET.

heavyrubberslave, 2024-11-02:
I have a question regarding the R27. According to this issue it should be "omitted". Does this mean I need to jump/connect the two pads of the R27 with some plain wire, or does it mean I don't need to solder anything and leave the pads for R27 completely blank?

CrashOverride85, 2024-11-02:
Yeah, just don't solder anything and leave the pad for R27 empty.
Without R27, I think the odds are good that a 2n7000 will be ok; it's working for me with an 2n7000 and R27 present, and removing R27 will only make it (much) more likely to work.
I'll _hopefully_ be posting an update about the redesigned SMD board in about a week.

closed 2024-11-06 by CrashOverride85