CrashOverride85/zc95 #135: Pairing shock collars

heavyrubberslave, 2025-01-07:
I ordered some shock collars that looked like the ones you show here but I cannot figure out how to pair them. I pressed the power button for 3s, it beeps, the light is blinking but what now? I tried to click TEST in the collar config menu several times but nothing happens. I haven’t paired them with the remote before, they‘re fresh out of the packaging. 

Can anyone maybe point out the obvious I‘m missing?

CrashOverride85, 2025-01-07:
That does sound like you're trying to pair them the correct way. BTW they can be re-paired anytime by pressing and holding power button for a couple of seconds as you describe, i.e. swapping them between remote(s) & zc95 isn't an issue.

The only things I can think of is 
1) double checking the 433 tx module is seated correctly and the right way around 
2) seeing if you can check what frequency the collar operate at, e.g. using an SDR.
Alternatively, remove the back of the remote, and see if it looks more like the 433 or 915 version:
https://github.com/CrashOverride85/collar/blob/master/extra/Type1.MD

heavyrubberslave, 2025-01-08:
Okay, good to know, thanks. Maybe the menu point "TEST" could then be renamed to "PAIR/TEST"? Just so it's obvious.

1. It is placed correctly. But I ordered 5 modules from AliExpress and one module even prevented the ZC95 from starting up and there was a high-pitched buzzing sound coming from the capacitors/diodes at the top of the mainboard. When I removed it, the device started up normally. Maybe the store I got the modules from was bad. Where did you get yours from?
2. I opened it and it looks 1:1 like the 433 version in your picture, except the PCB being blue instead of green. But the layout looks exactly the same.

CrashOverride85, 2025-01-09:
Yes, I agree renaming is probably a good idea. TBH I need to revisit shock collars soon, and might see if I can make the pairing process more obvious.

I can't remember where mine came from, but it would have either been aliexpress or ebay. TBH I've bought a few over the last few years from various sellers. But see #103 . it sounds like some of these modules are a bit iffy

heavyrubberslave, 2025-02-09:
I ordered new transmitter modules and with those the shock collars work. But the TEST function works randomly. Sometimes they vibrate every single time when I click "TEST", sometimes they just don't react at all for a few times pressing "TEST".

CrashOverride85, 2025-02-09:
I wonder if this is related to #111, I think I might need to increase the duration of the signal sent when test is pressed

heavyrubberslave, 2025-02-09:
Could be. Maybe it would be worth exploring the idea of making the signal length configurable per collar, so everyone can find the length that works best for their collar(s)?

CrashOverride85, 2025-02-10:
Yeah, good idea, having that configurable probably makes a lot of sense, will add it to the list