CrashOverride85/zc95 #137: Hardware check failed - Output 9v missing

simonsolutions, 2025-01-27:
I have assembled a complete box with all boards (Main, Output, Audio v0.4, 433Mhz Board) and the box worked. 

Now after one week not using it and reconnecting the battery the box shows FAULT on all channels and the Output Board is NOT lighting the LED 9V and OK - all other 3 are light.

- I assembled a new Output board and tried this. 
- I reflashed board Raspi Picos with the current and correct image. 
- I tried another battery.
Nothings changes and the FAULT message persists. 

Any idea what i can check further?

Thanks in advance.

CrashOverride85, 2025-01-27:
What's the exact error it's showing?
Also is the OK  LED off completely, or flashing? Does the 9v LED briefly come on when the box is first powered on, then go off? or is it never on?

Just to confirm, the output board isn't connected to USB power?

To get more info on what's going wrong, can you try connecting a ttl serial cable to the output boards serial header? (115200 baud)
On power up, it should be giving some info on what's going wrong.

simonsolutions, 2025-01-27:
The exact error I attached as photo.

[image]

The OK is completely off and the 9v LED never is on when turning on. 
Non of the Raspis or anything else is connected to USB power or something else. Just the battery.
This is on both output boards the same.

I tried a serial TTL, and didn't get any response. I just ordered another USB TTL converter to try.

CrashOverride85, 2025-01-27:
That's certainly a fault I've not encountered before. The 9v LED should always switch on moments after start up, then after a couple of seconds or so the OK LED should either stay on along with the 9v LED, or flash and the 9v LED switch off. 
As neither is happening, it suggests something is very wrong with the output board. 

The fault screen you're seeing does mean the output board has definitely been detected and successfully queried during startup though, which is odd given the lack of activity on the LEDs suggest it's not doing anything.

Could you try swapping the pico's over and reflashing? to rule out a problem with the pico on the output board.
Other than that, it's worth checking the actual voltages of the 5v & 12v supplies reaching the output board if you haven't already. A very low 12v would cause all channels to report a fault & possibly no 9V LED (but wouldn't explain no OK LED...)

simonsolutions, 2025-01-27:
Tthe exact same problem occurs with 2 completely separate output boards. so a mistake on one output board i would say could not be the reason. 
The LED behaviour with all voltage LED on and then seconds later the OK i had at the beginning. i dont know what happend during on week not using it and then reconnecting the battery. The first output worked already some time.

I reflashed the picos already with no change in behaviour. I ordered new ones (will arrive at the end of the week) to try new Picos. 

The voltages at the the measure points:
Battery directly: 12,66V (box turned off)
Output: TP1: 190 mV raising over time slowly to 266 mV
Mainboard:
TP1: 12,44 V
TP2: 0 V
TP3: 5,06 V
TP4: 3,346 V

CrashOverride85, 2025-01-27:
Yeah, two boards having the same issue does seem quite unlikely, but I'm struggling to see what else could cause it. Even if the self test fails, the OK LED should be flashing - that not happenings points to something being very wrong. 

The output board should only need +5v and ground to get a flashing OK led, and +12v as well to pass the self test. It should do that even if the main board isn't working (beyond providing power), e.g. including if the main board pico was absent.

Those voltages look absolutely fine - to rule out poor connections, could you also check them on output board connector on top of the PCB? +v5 and +12v are labelled on the silkscreen

simonsolutions, 2025-01-28:
I checked the +5v and +12v on the top of the output board and they have the same current as on the mainboard, so there are no breaks in this connection.

The interesting part that it worked and when replugging the battery after one week it showed this behaviour. The new picos are already shipped and should arrive by the end of the week. So maybe there is the problem in the pico itself. 

If it helps you I can send you a video of the behaviour via email.

CrashOverride85, 2025-01-28:
It does seem really odd that it initially worked, don't think I can explain that. I suspect getting serial output from it will help explain what's going on. Although hopefully a new pico fixes it 🤞

simonsolutions, 2025-01-28:
I tried serial output today, but didnt get any. Just to be sure on the output board i have to connect ground to ground of the 3,3 Serial TTL Converter und Tx of the board to Tx of the Converter? 115200 baud in the arduino ide serial monitor will work? or do i need something different?

As soon as the new Picos arrive i will test it an keep you posted. If it works then, we can see how we analyse the error in order to better address it in the future.

CrashOverride85, 2025-01-28:
ahhh, no, connect the TX on the output board to the RX line on the serial board (along with ground).
I've not used it with the Arduino serial monitor, but yes, that should work when set to 115200

simonsolutions, 2025-01-28:
Ok i'll try it Tx (board) <> Rx (converter). 
With which program did you log the serial output?

simonsolutions, 2025-01-29:
Ok I tried it but there is no output on the serial port with Gnd-Gnd and Tx-board<>Rx-converter.

I tried to remove the Pico on the Output Board. That changes the message in the display that the ZC624 is missing - what would be a correct behaviour. But the LEDs dont change at all. 9V and OK never light up and the other 3 light up.

CrashOverride85, 2025-01-29:
That is really confusing :/
As it's clearly being detected, it should be outputting to serial on startup. 
Could you double check with something else that the serial board is working as expected?
If it's not outputting to serial, there must be something wrong with the output board, even if it's affecting both of them. 
If the output board's pico is getting power, it outputs the firmware version to serial before any communication with the main board can happen.

simonsolutions, 2025-01-30:
Okay today the new Picos arrived, just taking a new Pico (non-W) soldering the pins, flashing the output firmware and now it works again. I dont know what the old pico had, but that seems now as it was the source of the problem.

CrashOverride85, 2025-01-30:
Nice 🎉 
At least it's working :)

simonsolutions, 2025-01-30:
Absolutely. If you need any analysis for further development, let me know

closed 2025-01-30 by simonsolutions